qca/open-plc-utils #40: Visual Studio 2010 - compile errors

lazna, 2015-02-12:
Just downloaded project zip file, unzip to local drive, browse to file 'plc-utils-2010.sln' doubleclick it, wait till project load into visual studio 2010 and than press F6. Many (more than 2000) various errors occurs than got 'Build failed' message. What am I doing wrong?

razzlefratz, 2015-02-13:
Hi Petr,

I tried the same thing on an 8 core Windows host and got loads of errors.
On further investigation, I discovered that Microsoft Visual Studio uses
the number of cores to determine the number of parallel projects to build
in parallel by default. I changed the number of parallel builds from 8 to 1
and 1only had a few projects not build. I need to look at them to determine
why those did not.

see
http://blogs.msdn.com/b/visualstudio/archive/2010/03/08/tuning-c-build-parallelism-in-vs2010.aspx
to see how to set the number of parallel builds.

Thanks for alerting me to this problem. I am not sure what to do about it
other than through documentation. I normally do not waste my time on
Windows.

Regards,
Charlie

lazna, 2015-02-13:
Hi Charles, thanks for reply.

Changed setting according link you sent to 1 Core, but does not help. Again more than 2000 error, in most it: "error C1083: Cannot open include file: '%filename%': No such file or directory". As far as I am not a programmer, looks to me like some path setting problem....

At the end of log, threre are also 4 following errors:

Error	2387	Could not find file 'E:\PLCutils\open-plc-utils-master\VisualStudioNET\\Programs\2010\plcwait.exe' 'Systém nemůže nalézt uvedený soubor.'	E:\PLCutils\open-plc-utils-master\VisualStudioNET\plc-utils\plc-utils-2010.vdproj	plc-utils

Could it be because this double backslash?

L.

razzlefratz, 2015-02-14:
I think you are right about it being a path problem. I need to investigate further and reason this through.

razzlefratz, 2015-02-14:
Petra,
I do not think the double slash will cause a problem. Try building program plcwait by itself and tell me what happens. If project plcwait.exe does build then the plc-utils package will not build either. I do not see that error message but I will keep looking around.

razzlefratz, 2015-02-16:
Petra,
I have successfully compiled all Visual Studio 2003 and 2010 projects on Windows 7 leaving the parallel builds at 1 for simplicity. Running parallel builds causes error messages to interleave, making them hard to read. There were some unresolved externals caused because we added or moved some modules without updating dependent project files. I think you may be good to go.

lategoodbye, 2015-02-16:
@lazna 

I got a lot of errors while running an anti-virus program during compilation. May be you could try to disable it.

lazna, 2015-02-16:
Thanks for your effort. Trying again: Cores are limited to 1, antivirus with turned off realtime scan and open-plc-utils folder is excluded, but still no luck. I am recorded my desktops action into video, maybe you will see some my general mistake...

http://www.uschovna.cz/en/zasilka/DC9XGW52ZA6DW8WJ-4YI//?set_lang=en

L.

lategoodbye, 2015-02-17:
Looking at the video shows me that 2 fundamental C header files (limits.h, stdio.h) from VS2010 cannot be found. These files belong to VS2010 and not open-plc-utils. Please try to fix this problem first.

lazna, 2015-02-17:
Do not know how should I fix this problem. Just performed "Repair installation" from setup menu, but it does not help. Finded some files with same names in my PC, try to copy them into VS 'include' directory, but this only produce same amount of different errors (cannot find some other files mentioned in this .h files). 
What I generally cant understand is, why there is not a binary distribution to download. AFAIK for program authors this is only one single click in VS, packing into zip and upload. VS is programmers tool and I do not understand why someone push non-programmers into suffering (time wasting) with software tool not suitable for non-programmers. I known Linux guys usually hate windows, but if is this case, why there is VisualStudio project present in source? Many opensource projects produce windows binaries to download, so why not here. No necessarily compile all versions, people like me will be happy for one compilation per year (to comprise possible new chipsets). Myabe I am wrong, but expect the development is financed by Intellon/Atheros/Qualcomm which is semiconductor producer, not a software retailer. What is difference between distributed SW in source and in binaries in this case??? I know, company is paranoid in its information publish limiting (remember from ath5k driver developers mailing list) but if there is an open source, than where the problem is? What about a little leak? ;-))) Someone can compile code and put binaries anonymously somewhere...
I want to buy more devices with atheros chips, this increase (not significantly, but anyway....) atheros profit. This is good motivation for your boss, isnt it??

Your, currently frustrated, customer.....

razzlefratz, 2015-02-17:
Hi Petr,

Thank you for sharing your frustration with me. I feel your pain. I will consider removing the Visual Studio project and solution files from the package. I have not had a chance to look at your video yet but will as soon as I have access to an open host.

Hang in there!

n1000, 2016-10-18:
I don't know if you are still around @lazna, but if so you may be interested in trying this branch, from pull request https://github.com/qca/open-plc-utils/pull/86 :

https://github.com/n1000/open-plc-utils/tree/update_msvc

lazna, 2016-10-20:
Hi, I am still aroud... Just downloaded zip file, bit found only plc-utils-2010.sln and plc-utils-2003.sln but you wrote you made it ready for Visual Studio 2015 ?? Mybe I miss something... Downloaded and installed VS2015 Community edition, but it says end of trial period :-/

N1000, please.... Why not publish windows binaries? In windows world there is no natural behaviour to compile programs before usage, non programmers (like me) could have continuous troubles with this issue. Are there any other problems (like licence) prevent to publish binaries? I can offer hosting for it if it is necessary.. Thanks

n1000, 2016-10-22:
Hi @lazna , registration for Visual Studio 2015 is free. I'll consider periodically releasing win32 binaries using the github releases feature.

n1000, 2016-10-22:
Closing this as the project now uses Visual Studio 2015 Community. See pull request #86.

closed 2016-10-22 by n1000