qca/open-plc-utils #42: GNU/Linux - compile errors

KlausKerzer, 2015-04-04:
Hi there, got the following compile errors:

../mme/ARPCPrint.c: In function ‘ARPCPrint’:
../mme/ARPCPrint.c:123:2: error: incompatible type for argument 3 of ‘vfprintf’
/usr/include/stdio.h:368:12: note: expected ‘__gnuc_va_list’ but argument is of type ‘void *’
Makefile:60: recipe for target 'ARPCPrint.o' failed
make[1]: *** [ARPCPrint.o] Error 1
make[1]: Leaving directory '/home/pi/dlan/open-plc-utils/plc'
Makefile:29: recipe for target 'all' failed
make: *** [all] Error 1

doublecheck.sh says:

ether
key
mdio
mme
ARPCPrint.c: In function ‘ARPCPrint’:
ARPCPrint.c:123:2: error: incompatible type for argument 3 of ‘vfprintf’
/usr/include/stdio.h:368:12: note: expected ‘__gnuc_va_list’ but argument is of type ‘void *’
nodes
nvm
pib
plc
In file included from amptest.c:135:0:
../mme/ARPCPrint.c: In function ‘ARPCPrint’:
../mme/ARPCPrint.c:123:2: error: incompatible type for argument 3 of ‘vfprintf’
/usr/include/stdio.h:368:12: note: expected ‘__gnuc_va_list’ but argument is of type ‘void *’
In file included from plctest.c:141:0:
../mme/ARPCPrint.c: In function ‘ARPCPrint’:
../mme/ARPCPrint.c:123:2: error: incompatible type for argument 3 of ‘vfprintf’
/usr/include/stdio.h:368:12: note: expected ‘__gnuc_va_list’ but argument is of type ‘void *’
ram
serial
tools

--

compiling on Raspbian, Raspberry Pi, fresh install, straight out of the box.

uname -a
Linux xxxxx 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Debian 4.6.3-14+rpi1' --with-bugurl=file:///usr/share/doc/gcc-4.6/README.Bugs --enable-languages=c,c++,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.6 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.6 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.6.3 (Debian 4.6.3-14+rpi1) 

make -v
GNU Make 4.0
Built for arm-unknown-linux-gnueabihf
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Thanks in advance,
Klaus

lategoodbye, 2015-04-05:
Did you tried the solution from #36 ?

n1000, 2015-09-26:
It does seem that the solution mentioned in issue #36 would resolve this. Please reopen if this is not the case.

closed 2015-09-26 by n1000