qca/open-plc-utils #44: They don't support the showing of 1000 mb+ speeds

Zero3K, 2015-05-10:
I'm helping out a guy that uses them as part of a script which offers a basic GUI for showing the status of a varitey of Homeplugs. While doing so, I have noticed that they incorrectly show the max speed of the Netgear PL1200s being only 128 mb and are not able to correctly show the speed of one of them that are on the network (one of them is showing a speed of 255 mb when it should be showing a speed of1060 mb or so). If anyone is interested in the script, you can get it by going to http://forum.ethernetovercoax.eu/index.php?topic=126.0.

razzlefratz, 2015-05-12:
Hi Zero3K

Your HomePlug Monitor is cool. Folks will like it.

You are correct in your observation. The INT6000, INT6300 and INT6400 use 8 bits in their VS_NW_INFO confirm message to express PHY rate because they cannot exceed 200 Mbs. Later chipsets, like the AR7400, QCA7410, QCA7450 and QCA7500, use 16 bits.

This difference creates the situation you are seeing when there are multiple chipset generations on the powerline network. I will try to explain the issues below and how to address them. 

Looking in the toolkit code, module NetworkInformation1.c sends a version 0 request and waits for a version 0 response while module NetworkInformation2.c sends a version 1 request and waits for a version 1 response. Function ReadMME, found in each of those modules, calls function UnwantedMessage to discard incoming messages until it receives the expected message type and message version. We use NetworkInformation1.c in int6k* programs and NetworkInformation2.c in amp* and plc* programs. The difference is the data structure used to define message payload. For simplicity I will call them the old format and new format.

Old firmware may not understand new requests and ignore them. Old devices running upgraded firmware will recognize old and new format and will respond accordingly. Newer, faster devices recognize old and new format and will respond accordingly; however, if a new device is transmitting faster than 255 Mbps it will truncate the speed to 255 Mbps to accommodate the older format. I believe that is what you are seeing and it is by design.

I am not sure what tools or what toolkit modules you are using to collect your HomePlug Monitor data but observe that older tools, like int6k and int6kstat, user old format VS_NW_INFO messages and newer tools, like amptool, ampstat, plctool and plcstat use new format VS_NW_INFO messages. There are not tools to handle all cases because of program complexity. I want all the tools to be simple so that folks can understand them.

The proper way to deal with this issue on a mixed network is to first send a VS_SW_VER request to the local device to determine its MAC address and device type so that you know what version of VS_NW_INFO request to send to it. Then send the VS_NW_INFO request to it in order to find the MAC addresses of other devices on the network. Then send a VS_SW_VER message to each device on the network to determine the device type of each so that you know what version of VS_NW_INFO message to send to each one in order to get their full PHY Rate. 

As an alternative, one could send a VS_SW_VER request to the broadcast address in order to identify the type of each network device but that method is not reliable when devices are loaded with traffic because vendor specific confirmations messages may get dropped under heavy load.

I hope this helps a bit.

Regards,
Charlie

Zero3K, 2015-05-12:
I'm not the maker of the script. I'm just a user of it.

n1000, 2020-09-26:
Closing stale issues that are not directly related to a bug in open-plc-utils.

closed 2020-09-26 by n1000