qca/open-plc-utils #47: Setting NMK on the local devices disregards device parameter

batmah, 2015-06-26:
Hi,
I have several ALLNET ALL168600 devices connected to same switch. I'd like to setup several independent networks with different NMKs on each of them.

~ # amptool -i eth2 -t 1000 -a
eth2 Fetch Device Attributes
eth2 MAC1 INT7400-MAC-5-3-5306-00-1402-20121214-FINAL-B (16mb) (0xF6) "60Hz" "Detected"  0x7F
eth2 MAC2 INT7400-MAC-5-3-5306-00-1402-20121214-FINAL-B (16mb) (0xF5) "50Hz" "Detected"  0x7F

When I am trying to change NMK on one of the local adapters with next command

amptool -i eth2 -t 1000 -K KEY1 -M MAC1

it reports
eth2 MAC1 Set Local Network Membership Key
eth2 MAC1 Setting ...

but all of my adapters are changing their NMKs together, not just MAC1 one. It looks like amptool does disregard device parameter. 
Could anyone look into it or explain what I am doing wrong?
Thanx.

batmah, 2015-06-26:
As I understand my devices are in Bootloader mode and all having the same 00:B0:52:00:00:01 MAC.
How could I have them leave this mode?

When I am trying to get nvm and pib files for ampboot tool I am receiving errors like:

$ amptool -i enp4s0 -e -n test.nvm -p test.pib MAC
enp4s0 MAC Read Firmware from Device
amptool: Bad Header Checksum: Operation canceled
enp4s0 MAC Read Parameters from Device
enp4s0 MAC Read test.pib

Any suggestions?

batmah, 2015-06-29:
If anybody interested, for me next patch did the job

--- SetNMK.c.original	2015-06-29 12:01:49.000000000 +0200
+++ SetNMK.c	2015-06-29 11:53:55.127479562 +0200
@@ -103,7 +103,7 @@
 #endif
 
 	memset (message, 0, sizeof (* message));
-	EthernetHeader (&request->ethernet, localcast, channel->host, channel->type);
+	EthernetHeader (&request->ethernet, channel->peer, channel->host, channel->type);
 	QualcommHeader (&request->qualcomm, 0, (VS_SET_KEY | MMTYPE_REQ));
 	plc->packetsize = sizeof (struct vs_set_key_request);
 	request->EKS = 0x01;

n1000, 2016-10-18:
Merged, thanks.

closed 2016-10-18 by n1000

ginjo, 2019-01-15:
Hi @batmah ,

Before I post this as a new issue, I wanted to check in with you...

Did you ever figure out how to solve the ```Bad Header Checksum``` issue? I'm tryng to extract firmware with ```int6k```, but I get the same error as you reported above.

```
$ int6k -n old.nvm MAC
en0 <MAC> Read Firmware from Device
int6k: Bad Header Checksum: Operation canceled
```

I've also tried this with the ```plctool``` command, which returns a different error:

```
$ plctool -e -n old.nvm MAC
en0 <MAC> Read Module from Memory
en0 <MAC> Module not available in NVM or Memory (0x32): Device refused request
```

Either way I have not been able to extract firmware from any of my devices. Google turns up nothing for these errors, so maybe I'm doing something wrong or missing some intermediate step.

My hosts are a mix of:

OSX (El Cap) with
```int6k open-plc-utils-0.0.6 OSRQCA-929 ISO C Qualcomm Atheros, San Jose CA, USA```

OpenWRT (BB) with
```int6k open-plc-utils-0.0.2 OSR-6599 ISO C Qualcomm Atheros, Ocala FL USA 08:35:33 Sep 23 2014```

My devices are all some flavor of the following:
```
INT6400 INT6000-MAC-4-4-4405-05-4514-20110819-FINAL-C
QCA6410 MAC-QCA6410-1.1.0.852-03-20130128-FINAL
```

BrynlyClarke, 2021-02-19:
Did you get anywhere with this?  I am just starting to try understand some of these tools, to try and rescue some Solwise HomePlug APs I think are “soft-bricked” - I get exactly the same Int6k Header Checksum error, on a device I know is working correctly

batmah, 2021-02-19:
Sorry, I am not working with those devices any more.

BrynlyClarke, 2021-02-19:
@ginjo Did you get anywhere with this? I am just starting to try understand some of these tools, to try and rescue some Solwise HomePlug APs I think are “soft-bricked” - I get exactly the same Int6k Header Checksum error, on a device I know is working correctly

ginjo, 2021-02-23:
Sorry @BrynlyClarke , I was never able to extract the firmware from any of these devices. However, I am using the devices fairly successfully in several locations. The only recurring issue I have is an occasional soft-crash (maybe your soft-brick?). I'm able to recover/reboot those devices (even remotely!) with a ```plctool``` command. You just need to know the NMK for your devices and the DAK of the crashed device.

One thing that may-or-may-not be helpful is some python code written to demonstrate security vulnerabilities in the plc protocol these devices use (or more accurately, in vendor's implementation of the protocol).  https://github.com/sogeti-esec-lab/HomePlugPWN