qca/open-plc-utils #49: Unable to flash firmware into QCA7000 with plcboot

lategoodbye, 2015-10-22:
The current version of plcboot is not able to flash a firmware permanently into the QCA7000.

    > ./plcboot -S ~/Desktop/NvmSoftloader-QCA7000.nvm -N ~/Desktop/MAC-QCA7000.nvm -P ~/Desktop/QCA7000.pib -F
    eth0 00:B0:52:00:00:01 Write /home/user/Desktop/MAC-QCA7000.nvm (1) (00000040:7936)
    eth0 00:B0:52:00:00:01 Start /home/user/Desktop/MAC-QCA7000.nvm (1) (000000C0)
    eth0 00:B0:52:00:00:01 Write /home/user/Desktop/QCA7000.pib (1) (00200000:9040)
    eth0 00:B0:52:00:00:01 Write /home/user/Desktop/MAC-QCA7000.nvm (5) (002C0960:308788)
    eth0 00:B0:52:00:00:01 Start /home/user/Desktop/MAC-QCA7000.nvm (5) (002C0D50)
    eth0 00:B0:52:00:00:01 MAC-QCA7000 is running
    eth0 00:B0:52:00:00:01 Start Module Write Session
    eth0 00:B0:52:00:00:01 No NVM Softloader Present in Flash Memory (0x71): Device refused request

I've bisect this issue to this commit 7f08748 .

ch-aurich, 2015-10-23:
@razzlefratz this was your commit. Before I merge Stefans patch I would make sure that we understood the issue correctly. 

I would expect that the bug causes plcboot to not work for all QCA PLC chips. Can you verify this assumption?

ch-aurich, 2015-11-03:
We are now quite sure to have understood this correctly - I therefore merged lategoodbye's changes

closed 2015-11-03 by ch-aurich