qca/open-plc-utils #67: Support for Broadcom HomePlug chipsets?

shmerl, 2016-02-15:
Sorry if it's a wrong place to post this. Does open-plc-utils support Broadcom chipset based HomePlug adapters like ZyXEL PLA5456KIT HomePlug AV2? And if not, are there any plans to add such support?

n1000, 2016-02-15:
open-plc-utils mainly focuses on Qualcomm Atheros Vendor Specific Management Messages. As far as I know there are no plans to add Broadcom chipset support to this project.

lategoodbye, 2016-02-21:
@shmerl AFAIK there is no public documentation about the BCM60500 so it would be very hard to implement.

n1000, 2016-10-18:
Closing this for now, we could reconsider this if the vendor specific MME information is released publicly.

closed 2016-10-18 by n1000

mirh, 2024-05-10:
https://github.com/serock/pla-util