qca/open-plc-utils #72: Windows version of the toolkit is missing?

Martin-P, 2016-05-24:
Hi all,

The docbook says the following in the chapter [Toolkit Installation on Windows 7™](https://github.com/qca/open-plc-utils/blob/9ddb713cc0ad5ce496c9e46e671133e364377b41/docbook/ch01s07.html#L8):

> The Windows™ version of the toolkit comes pre-compiled and ready to install and run using the native Windows Installer. You may copy and distribute installation file plc-utils-x.x.x.msi.

Can you please tell me where I can find these files? There are some references to FTP locations, but none of those work (anymore).

lategoodbye, 2016-05-24:
Hi @Martin-P,
the open-plc-utils doesn't have a Windows Installer, only the closed source variant has. This documentation issue needs to be fixed.

Martin-P, 2016-05-30:
Thanks for the info @lategoodbye,
do you know where I can find more detailed information about the closed source variant? I [searched](https://www.google.nl/search?q=site:www.qualcomm.com+powerline+toolkit) the Qualcomm site and unfortunately can't find any info about a powerline toolkit.

ch-aurich, 2016-06-07:
AFAIK the powerline toolkit is only available from Qualcomm under NDA, please contact you local sales rep if you need more information.

lategoodbye, 2016-06-10:
Btw It's possible to compile most of the tools with Visual Studio Community 2015.

Martin-P, 2016-06-29:
Thanks @ch-aurich and @lategoodbye for your help and info!

closed 2016-07-12 by n1000