qca/open-plc-utils #73: amptone not working with QCA7500?

kultate, 2016-06-15:
Hi,

I have been using amptone succesfully to measure signal quality with QCA7420 based adapters. I would like to make similar measurements with a QCA7500 based adapter, but amptone gives me an error:

$ amprate -i eth0 -rn
eth0 00:B0:52:00:00:01 Request Version Information
eth0 F4:06:8D:AD:2C:09 QCA7500 MAC-QCA7500-2.2.1.770-00-20160307-CS
eth0 F4:06:8D:AD:2C:09 F4:06:8D:AD:2B:48 TX 611 mbps Alternate
eth0 F4:06:8D:AD:2C:09 F4:06:8D:AD:2B:48 RX 741 mbps Alternate
$ amptone -i eth0 -s F4:06:8D:AD:2C:09 F4:06:8D:AD:2B:48
amptone: Too many carriers: Value too large for defined data type
$ 

Is there something that I could do to make amptone work with QCA7500 chipset?

Thanks!

lategoodbye, 2016-06-25:
Hi @kultate,
you could try plctone but it's possible this chipset isn't implemented yet.

lategoodbye, 2016-08-15:
Okay, i can confirm that ToneMaps2 and SignalToNoise2 can't handle the QCA7500. The MME response seems to be different.

@n1000 Any plans to fix this?

n1000, 2016-08-28:
I don't have the documentation that covers the new PIB / Prescalar format for the QCA7500. I'll look into what it would take to make this information available.

n1000, 2016-10-18:
I was apparently confused when I posted my last comment, as this question is not about PIB / Prescalar format, but rather about the SNR / modulation rate per subcarrier. I'll have to look into this further.

dchard, 2017-09-22:
I have the same problem. Tried with a TP-Link PA8010P (QCA7500), and both amptone and plctone is not able to read due to " Too many carriers: Value too large for defined data type"

Interesting is,  int6 tool can read various parameters, and can read tones up to 916 tones:

```
root@powerline:/home/thlab# int6ktone -i ens33 EC086B86B531 EC086B86B530 -s
 SNR    3.996   17.012   17.396    3.996    3.996    3.996    8.398
 ATN  -56.004  -42.988  -42.604  -56.004  -56.004  -56.004  -51.602
 BPC    1.998    5.978    6.065    1.998    1.998    1.998    3.339
 AGC 00 68 68 00 00 00
 GIL 00 68 68 00 00 00
```

```
root@powerline:/home/thlab# int6ktone -i ens33 EC086B86B531 EC086B86B530 -h
 000 02 00 00 02 02 02 002 ##
 001 02 00 00 02 02 02 002 ##
 002 02 00 00 02 02 02 002 ##
 003 02 00 00 02 02 02 002 ##
 004 02 00 00 02 02 02 002 ##
 005 02 00 00 02 02 02 002 ##
 006 02 00 00 02 02 02 002 ##
 007 02 00 00 02 02 02 002 ##
 008 02 00 00 02 02 02 002 ##
 009 02 00 00 02 02 02 002 ##
 010 02 00 00 02 02 02 002 ##
 011 02 00 00 02 02 02 002 ##
 012 02 00 00 02 02 02 002 ##
 013 02 00 00 02 02 02 002 ##
 014 02 00 00 02 02 02 002 ##
 015 02 00 00 02 02 02 002 ##
 016 02 03 03 02 02 02 005 #####
 017 02 03 03 02 02 02 005 #####
 018 02 03 03 02 02 02 005 #####
 019 02 03 03 02 02 02 005 #####
 020 02 06 06 02 02 02 011 ###########
 021 02 06 06 02 02 02 011 ###########
 022 02 06 06 02 02 02 011 ###########
 023 02 06 06 02 02 02 011 ###########
 024 02 10 12 02 02 02 021 #####################
 025 02 10 12 02 02 02 021 #####################
 026 02 10 12 02 02 02 021 #####################
 027 02 10 12 02 02 02 021 #####################
 028 02 10 10 02 02 02 019 ###################
 029 02 10 10 02 02 02 019 ###################
 030 02 10 10 02 02 02 019 ###################
 031 02 10 10 02 02 02 019 ###################
 032 02 04 04 02 02 02 008 ########
 033 02 04 04 02 02 02 008 ########
 034 02 04 04 02 02 02 008 ########
 035 02 04 04 02 02 02 008 ########
 036 02 00 00 02 02 02 002 ##
 037 02 00 00 02 02 02 002 ##
 038 02 00 00 02 02 02 002 ##
 039 02 00 00 02 02 02 002 ##
 040 02 00 00 02 02 02 002 ##
 041 02 00 00 02 02 02 002 ##
```
Would be nice to make this work. There are quite a lot of powerline devices out there with this chipset.