qca/open-plc-utils #74: Bridged Network (brctl) and AR7400

DoPeRaSo, 2016-06-23:
Hello,
I own a DHP-1565 and installed OpenWRT (custom build with a patch from http://www.fredericb.info/2016/02/powerline-plc-support-in-openwrt-for-d.html to enable the communication to the A7400) and I am able to load the blob and communicate - but only if I did not attach this last port to the ethernet bridge and let it be a "single" interface - as soon, as I connect it to the bridge, I am not able to load the blob again (so in fact I need do "brctl delif ...", load pid and nvm, and later I have to attach the interface again)

The problem is, that the neither amphost nor amptool seem to be able to communicate with the chip if the port of the PLC adapter (eth0.3) is contained within the bridge:
I only get this three lines:
# /overlay/upper/open-plc-utils/amptool -i eth0.3 -Iar
eth0.3 00:B0:52:00:00:01 Request Version Information
eth0.3 00:B0:52:00:00:01 Fetch Device Attributes
eth0.3 00:B0:52:00:00:01 Device Identity

I'm not sure where the problem originates: either OpenWrt (brctl?) or the Open-PLC-Utils and I jsut dont't know, how to solve the issue.

Can someone give me an advice?

Best regards,
Peter

dvlemplgk, 2016-06-24:
Hi

MME frames sent by the PLC are not received on eth0.3 but on the bridge interface.
You could install some ebtables rules so that the PLC's MME frames are not bridged and can be read by plctools from eth0.3 again.

	ebtables -t broute -A BROUTING -p 0x88e1 -s 00:B0:52:00:00:01 -j DROP
 	ebtables -t broute -A BROUTING -p 0x88e1 -s xx:xx:xx:xx:xx:xx -d Broadcast -j DROP
 	ebtables -t broute -A BROUTING -p 0x88e1 -d yy:yy:yy:yy:yy:yy -j DROP

replace xx:xx:xx:xx:xx:xx with the mac address of PLC configured in the PIB, yy:yy:yy:yy:yy:yy with the mac address of eth0.3.
The first rule is for booting the PLC chip, second and third for management after PLC boot.
A single rule for all MME frames should also work:
         ebtables -t broute -A BROUTING -p 0x88e1 -j DROP

Regards
Günther

n1000, 2017-05-13:
Assuming this was resolved, please reopen as needed.

closed 2017-05-13 by n1000