qca/open-plc-utils #84: Windows binaries?  MME docs?

unknown14725, 2016-10-11:
I cannot find anywhere else to ask, except sending emails to the addresses found in this project, and I don't want to do that. (yet)

Are there any windows binaries of this project anywhere? I've spent the last couple of weeks googling like a maniac, trying to find similar software, or even documentation describing the Qualcomm Atheros MME protocol. I know how the Homeplug AV protocol works, so it's the Vendor MME specific part I need info on.

I'm trying to code my own small console program to get some basic network stats from my TP-LINK devices, but it's ridicolous hard without having any kind of documentation on the Vendor MME protocols.
The source files found here keep referring to all kinds of technical manuals and books, but they are nowhere to be found online. So far I've studied the source files here, with the help from wireshark, trying to figure out stuff... but it's becoming apparent that it takes too much effort for what its worth.

So I'm in the process of giving up, and hoping that someone has windows binaries for this project so I don't have to agonize over this even more. Or can provide links to some documentation describing the MME.

Thanks in advance.

n1000, 2016-10-15:
Do you know what PLC chipset is in your TP-LINK device? If you have an older chipset, you can find Win32 compiled versions of some tools here:

http://support.dlink.com/ProductInfo.aspx?m=DHP-306AV

Check out the "Powerline Utility 2.13" installer. You don't actually have to install it, the exe can be opened with 7zip to get the int6k* tools.

unknown14725, 2016-10-16:
Hey there. 

I had a look at the tools in your provided link, but they are for different hardware. Even so, they do not work. I'm looking in Wireshark to inspect the packets they'd send, but they aren't even generating any outgoing traffic. At all. Not even doing broadcasts with the tool generate any outgoing packets.

I have several kits labelled TP-Link TL-PA411KIT, which seem to be paired AV500 (EU V2.0) devices.

0xA000 to my devices returns this CNF:
```
Device ID: QCA7450/QCA7420 (0x20)
Version: MAC-QCA7420-1.0.0.337-01-20120309-FINAL
```
0xA054 return this CNF:
`Manufacturing String: Qualcomm Atheros HomePlug AV Device`

0xA068 return this CNF:
```
Hardware platform: QCA7420
Software platform: MAC
Major version: 1
Minor version: 1
Software/PIB version: 20120309
Software build number: 0
```

I see "Qualcomm Atheros Powerline Firmware Technical Reference Manual" and "Qualcomm Atheros HomePlug AV Firmware Technical Reference Manual" mentioned several times in the source files. Do you know where these are available? The Quallcomm website has no resources whatsoever regarding HomePlug AV or related technologies.

Any help would be much appreciated. As you can see above, I'm able to retrieve _some_ info from my devices (using AutoIt and WinPcap dll), but I'd love to learn how to do more stuff with them.

n1000, 2016-10-16:
As far as I know, none of the documents you mentioned have been released publicly, and they are only available via NDA.

Regarding the tools I provided the link for, be sure to specify the interface that you wish to use:

int6k -i N -r

Where N is an integer starting at 0. Unfortunately, the tool used to find out what interface number to use (pcapdevs.exe) is not included in the package I mentioned, so you will have to guess which number to use. On my windows system it happens to be "-i 2". Unless you have a huge number of interfaces on your machine, it will most likely be a number below 5.

I just double checked this on my Windows machine (Windows 10), and I'm able to see the management frames going out in Wireshark.

n1000, 2016-10-16:
One other thing, I have been tinkering with cross compiling these tools for windows using mingw32 (from Linux / OpenBSD). I can probably provide a full set of binaries later.

n1000, 2016-10-16:
Just built this using the mingw32 branch I mentioned above:

https://www.dropbox.com/s/co3xkp1iqqeat2d/win32-bin-d071b109.7z?dl=0

It is all of the tools as of git commit d071b109ba6410dbcc83e347c6af20d570b82282 .

I have not extensively tested these, but at least int6k -r worked on my system.

unknown14725, 2016-10-17:
Oh wow, thank you!

I didnt realize the software wasn't able to determine the active NIC by itself. I tried using the -i argument

`int6k -m -i 1 all`

...and now it works. I also downloaded your compiled toolset, and they work like a charm. Finally I can get some real work done in Wireshark figuring out how to format my packets properly. 

Once again, thanks. I really appreciate your help.

n1000, 2016-10-17:
Sure, no problem. Maybe you can check out the branch below, I updated the project to use visual studio 2015 community edition. This should just work out of the box with no tweaks required (hopefully):

https://github.com/n1000/open-plc-utils/tree/update_msvc

Closing this issue now. If you try the above and find any issues, or even if it just works, drop a comment on this pull request:

https://github.com/qca/open-plc-utils/pull/86

closed 2016-10-17 by n1000

fernando7650, 2019-05-31:
Thank you so much!! I could recover my bricked TP Link device on windows :)