qca/open-plc-utils #88: Can't change DAK on QCA7000

mbtronics, 2016-10-26:
I'm trying to change the DAK on a QCA7000 with modpib and plctool, but it does not change. All other settings I try to change work. Any ideas?

--------------------------------------------------

$ modpib -D 09:C5:AB:AD:F9:FF:2C:5E:D8:A1:9D:CF:BB:E9:65:80 -v QCA7000-SpiSlave-HomePlugGP_CE-ClassB.pib
        PIB 0-0 0 bytes
        MAC 00:05:FE:44:A5:A7
        DAK 09:C5:AB:AD:F9:FF:2C:5E:D8:A1:9D:CF:BB:E9:65:80
        NMK 50:D3:E4:93:3F:85:5B:70:40:78:4D:F8:15:AA:8D:B7 (HomePlugAV)
        NID B0:F2:E6:95:66:6B:03
        NET Qualcomm Atheros Enabled Network
        MFG Qualcomm Atheros HomePlug AV Device
        USR Qualcomm Atheros Enabled Product
        CCo Auto
        MDU N/A

$ plctool  -P QCA7000-SpiSlave-HomePlugGP_CE-ClassB.pib 
eth1 00:B0:52:00:00:01 Start Module Write Session
eth1 00:B0:52:00:00:01 Flash QCA7000-SpiSlave-HomePlugGP_CE-ClassB.pib
...
eth1 00:B0:52:00:00:01 Close Session
eth1 00:B0:52:00:00:01 Reset Device
eth1 00:05:FE:44:A5:A7 Resetting ...

# plctool -I
        PIB 0-0 8660 bytes
        MAC 00:05:FE:44:A5:A7
        DAK 68:9F:07:4B:8B:02:75:A2:71:0B:0B:57:79:AD:16:30 (HomePlugAV)
        NMK 50:D3:E4:93:3F:85:5B:70:40:78:4D:F8:15:AA:8D:B7 (HomePlugAV)
        NID B0:F2:E6:95:66:6B:03
        NET Qualcomm Atheros Enabled Network
        MFG Qualcomm Atheros HomePlug AV Device
        USR Qualcomm Atheros Enabled Product
        CCo Auto
        MDU N/A

ch-aurich, 2016-10-26:
AFAIK the DAK cannot be reprogrammed. This means that you can only write it via bootloader mode where firmware, softloader and PIB are written to the flash initially. So what you observes is expected behaviour.

closed 2016-10-26 by ch-aurich

mbtronics, 2016-10-27:
Didn't know that, thanks!

ch-aurich, 2016-10-30:
Yes, you can - on the QCA7000 and QCA7500 chips I know this is done via bootstrap pins. That means that you must know the pin in hardware that is used to configure the boot source.
As the information given to me by Qualcomm is covered by an NDA I am not able to help you on this issue.
And even if you would know what pin to configure for bootloader mode you would not have the firmware and softloader to programm into the flash.

mbtronics, 2016-10-30:
Thanks for the info, I was afraid this would be te only way to do it.
(We have all information and firmware from Qualcomm under NDA, so that is not a problem.)