qca/open-plc-utils #103: compiling problems on raspbian

jacmendt, 2017-04-11:
i try to compile the open-plc-utils package on a  `Raspbian GNU/Linux 8` (armv61) and run into the following error with `make install`

```bash
make[1]: Entering directory `/home/pi/open-plc-utils-ether`
gcc -o efbu efbu.o getoptv.o putoptv.o version.o error.o todigit.o hexencode.o uintspec.o hexdump.o channel.o openchannel.o sendpacket.o readpacket.o closechannel.o basespec.o 
/usr/bin/ld: cannot open output file efbu: Permission denied
Makefile:52: recipe for target `efbu` failed
make[1]: *** [efbu] Error 1
make[1]: Leaving directory `/home/pi/open-plc-utils-ether`
Makefile:29: recipe for target `install` failed
make: *** [install] Error 1
```

I tried to compile the current master branch. Any advice?

Kind regards, JM

mhei, 2017-04-11:
Can you provide your full sequence of commands after git clone?

If you don't mind, you could also try my pre-compiled binaries from: https://packagecloud.io/mhei/open-plc-utils

jacmendt, 2017-04-11:
i did a git clone and executed a `make install` in the root directory of the project as root user like described in the [readme.md](https://github.com/qca/open-plc-utils/blob/master/README).

Your pre-compiled binaries worked for me. So thanks for this.

mhei, 2017-05-01:
Just for the record: I tried to reproduce your issue on my Raspberry Pi but without luck. Only when I "forget" to use _sudo_ I see the same error message which is expected behaviour. Anyway, the README should be slightly adjusted, so I'm filing a PR for this.

n1000, 2017-05-13:
I'm also able to build without issue on raspbian. Please re-open this bug and provide additional details if you are still seeing this issue.

commit 0f8c1ff4 (2017-05-21): Merge pull request #107 from mhei/issue-103

README: split make process into two steps (fixes #103)
commit f485cbaa (2017-05-21): README: split make process into two steps (fixes #103)

The compilation itself should normally be run without elevated privileges,
so split it from the install step.

While at, adjust some minor formatting issues.
closed 2017-05-13 by n1000