qca/open-plc-utils #111: FORWARDING_STA value always 0 in evse_cm_slac_param.c ?

bonjour81, 2017-11-07:
Hello, 
According to some comments in evse_cm_slac_param.c, " _EVSE-HLE may send zeros for the forwarding station since it is not used here_" .

Also in https://github.com/qca/open-plc-utils/blob/master/slac/evse.1

it is mentionned:
_Controls whether the msound recipient sends signal attenuation characteristic profile data to the EVSE-HLE or another station.
The only response specified at this time is the EVSE-HLE.
The default value is \fB0\fR.
Any other value is ignored._

So I understood evse util will always send 00:00:00:00:00:00 in there.

In the mean time, in the ISO15118:3  A.9.1.2 [V2G3-A09-04]  (2015-05), the content of CM_SLAC_PARM.CNF is discribed this way:

_FORWARDING_STA: The destination of SLACresults is always the EVHost_

So, If I understood correctly, we should have the EV address here instead of 00:00:00:00:00:00

pedrorivera, 2020-06-05:
Hi,

I have the exact same question. @bonjour81 did you figure it out?

Some other docs say FORWARDING_STA can be ignored if the RESP_TYPE field is 0x0. However, the table from the standard you reference is saying it's should be fixed to 0x1.  My assumption right now is the creators of this SLAC demo code decided to leave out that part, which is fine for this purpose because they could have also chosen to ignore it on the opposite end (i.e. pev.c). 

Would a real ECU require RESP_TYPE to be 0x1?

pedrorivera, 2020-07-08:
For completeness, I have tested a real ECU and it did require FORWARDING_STA to be the EV host address and RESP_TYPE to be '1'. Achieving this behavior was trivial, only involved introducing the following code below line 107 in evse_cm_slac_param.c:

`memcpy (session->FORWARDING_STA, request->ethernet.OSA, sizeof (session->FORWARDING_STA));`

And obviously change the response type to '1' in the evse.ini file.

n1000, 2020-10-07:
Fixed by pull request https://github.com/qca/open-plc-utils/pull/134

closed 2020-10-07 by n1000