qca/open-plc-utils #120: plctool  and/or evse  issue the message "memory after memcpy"

MoeSzyslak3, 2018-10-17:
With each use plctool.exe and evse.exe work as usual but both programs report "memory after memcpy" followed by series of hex numbers. See the full message below: 

memory after memcpy
FFFFFFF8 FFFFFFCA FFFFFFB8 43  06 75 00 01  FFFFFF87 08 21 4C  FFFFFF88 FFFFFFE1 00 01

Both programs seem to work proper. But the PLC link became more unstable. 

What is the reason for this behavior and how can I solve the problem?

mhei, 2018-10-17:
When I grep through the open-plc-utils source code, I do not find this or any similar message.
Are you sure, you are using latest git version? Maybe this message is triggered by libpcap?
Please provide a full log of your session, including full command line parameters, full output etc. Please also describe how you built open-plc-utils. For my quick test, I used artefacts built by appveyor of @wherget and installed winpcap 4.1.3 on a Win10 VM.

MoeSzyslak3, 2018-10-18:
Sorry for wasting your attentition. I totally forgot that I have inserted the code sequence causing the described message by myself in the local copy of ..\ether\readpacket.c.  

And you are right. I enabled WINPCAP. The reason is that data to process are transmitted via an ethernet link.

closed 2018-10-18 by MoeSzyslak3