qca/open-plc-utils #122: FreeBSD

Toutzn, 2018-10-24:
Hi,

Is it possible to extend the program for FreeBSD?

Thanks a lot. 

Regards

Torsten

n1000, 2018-10-25:
Hi Torsten,

I don't have access to a FreeBSD machine right now, but open plc utils does currently support OpenBSD and Mac OS. Porting it over should be pretty straightforward, since they all support the BPF interface.

Thinking about it more, I'm not sure why we didn't already add FreeBSD support already... No promises, but I may try this soon.

Thanks,
Nate

n1000, 2018-10-25:
Hi Torsten, please see the above pull request which adds FreeBSD support. I did not do extensive testing, but I was able to at least request the device identity ("int6k -i em0 -r all" in my case), and get a correct response.

I tested on FreeBSD 11.2. You will need to edit the top level "make.def" and replace instances of "gcc" with "clang", unless you have also installed gcc on your FreeBSD machine. "gmake" should also be used ("sudo pkg install gmake" if you don't already have it).

Let me know if this works for you.

-Nate

Toutzn, 2018-10-25:
Hi Nate.

Taks a lot for your reply. I will test it today and give feedback to you.

Torsten

Toutzn, 2018-10-25:
Hi Nate.
I tested your changes on FreeBSD11.2 wit gcc and clang and got following errors:

root@FREEBSD_11-2:~/open-plc-utils # make
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c efbu.c -o efbu.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/getoptv.c -o getoptv.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/putoptv.c -o putoptv.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/version.c -o version.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/error.c -o error.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/todigit.c -o todigit.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/hexencode.c -o hexencode.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/uintspec.c -o uintspec.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/hexdump.c -o hexdump.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c channel.c -o channel.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c openchannel.c -o openchannel.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c sendpacket.c -o sendpacket.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c readpacket.c -o readpacket.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c closechannel.c -o closechannel.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_   -c ../tools/basespec.c -o basespec.o
gcc -Wall -Wextra -Wno-unused-parameter -DMAKEFILE -D__GETOPT_H__ -D_GETOPT_DEFINED_     efbu.c  -o efbu
/tmp//ccI8w5eN.o: In function `function':
efbu.c:(.text+0xfc): undefined reference to `error'
efbu.c:(.text+0x11f): undefined reference to `sendpacket'
efbu.c:(.text+0x157): undefined reference to `error'
/tmp//ccI8w5eN.o: In function `main':
efbu.c:(.text+0x211): undefined reference to `channel'
efbu.c:(.text+0x22a): undefined reference to `channel'
efbu.c:(.text+0x23e): undefined reference to `channel'
efbu.c:(.text+0x24a): undefined reference to `channel'
efbu.c:(.text+0x285): undefined reference to `uintspec'
efbu.c:(.text+0x293): undefined reference to `channel'
efbu.c:(.text+0x29c): undefined reference to `channel'
efbu.c:(.text+0x2b0): undefined reference to `channel'
efbu.c:(.text+0x2b5): undefined reference to `hexencode'
efbu.c:(.text+0x2e5): undefined reference to `error'
efbu.c:(.text+0x303): undefined reference to `basespec'
efbu.c:(.text+0x30a): undefined reference to `channel'
efbu.c:(.text+0x315): undefined reference to `channel'
efbu.c:(.text+0x31e): undefined reference to `channel'
efbu.c:(.text+0x32e): undefined reference to `channel'
efbu.c:(.text+0x349): undefined reference to `uintspec'
efbu.c:(.text+0x354): undefined reference to `channel'
efbu.c:(.text+0x35d): undefined reference to `channel'
efbu.c:(.text+0x378): undefined reference to `uintspec'
efbu.c:(.text+0x383): undefined reference to `channel'
efbu.c:(.text+0x38c): undefined reference to `channel'
efbu.c:(.text+0x3ab): undefined reference to `getoptv'
efbu.c:(.text+0x3f9): undefined reference to `error'
efbu.c:(.text+0x3fe): undefined reference to `channel'
efbu.c:(.text+0x403): undefined reference to `openchannel'
efbu.c:(.text+0x429): undefined reference to `channel'
efbu.c:(.text+0x433): undefined reference to `channel'
efbu.c:(.text+0x438): undefined reference to `closechannel'
collect2: error: ld returned 1 exit status
*** Error code 1

Stop.
make[1]: stopped in /root/open-plc-utils/ether
*** Error code 1

Stop.
make: stopped in /root/open-plc-utils

mhei, 2018-10-25:
Please use gmake instead of make.

Toutzn, 2018-10-25:
Ok. at least one question:
if interface name is igb or vlan, not em. Does that make a difference, or does it matter?
Result:
root:~ # int6k -i igb1 -r all
igb1 FF:FF:FF:FF:FF:FF Request Version Information
root:~ # int6k -i igb0 -r all
igb0 FF:FF:FF:FF:FF:FF Request Version Information
root:~ # int6k -i igb2 -r all
igb2 FF:FF:FF:FF:FF:FF Request Version Information
root:~ # int6k -i vlan170 -r all
vlan170 FF:FF:FF:FF:FF:FF Request Version Information

n1000, 2018-10-25:
Hi Torsten. Can you give this a try on an Ethernet interface without any VLAN tagging enabled? I don't think VLAN tagging interfaces are fully supported yet.

On my system I have the following ifconfig output (I deleted my "inet6" lines, and my loopback (lo0) interface):

```
$ ifconfig | grep -e '^[a-z]' -e vlan -e options
em0: flags=8943<UP,BROADCAST,RUNNING,PROMISC,SIMPLEX,MULTICAST> metric 0 mtu 1500
        options=9b<RXCSUM,TXCSUM,VLAN_MTU,VLAN_HWTAGGING,VLAN_HWCSUM>
        nd6 options=23<PERFORMNUD,ACCEPT_RTADV,AUTO_LINKLOCAL>
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> metric 0 mtu 16384
        options=600003<RXCSUM,TXCSUM,RXCSUM_IPV6,TXCSUM_IPV6>
        nd6 options=21<PERFORMNUD,AUTO_LINKLOCAL>
em0.5: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 1500
        options=3<RXCSUM,TXCSUM>
        nd6 options=21<PERFORMNUD,AUTO_LINKLOCAL>
        vlan: 5 vlanpcp: 0 parent interface: em0
        groups: vlan
```
For me, the int6k tool is working fine on the em0 interface, but not em0.5 virtual interface. em0 is not configured to apply VLAN tags at all (I confirmed it with tcpdump), but em0.5 is having VLAN tags applied.

Are vlan tags being applied for all of your interfaces? you should be able to run the command above to  find out...

A couple other questions:

1. Can you confirm the interfaces are in the "UP", and "RUNNING" state according to ifconfig output? On BSD, generally an error will not occur if operation is attempted on an interface that is not in the "UP" state, but all packets will be silently dropped (at least that was my previous experience in OpenBSD).

2. What is the architecture of your FreeBSD machine? Is it just a standard amd64 / intel 64 bit CPU? (worried about endianness issues)

mhei, 2018-10-25:
@n1000: I quick-tested your branch on a FreeBSD 12.0-BETA1 amd64 VM and it looks good.

Toutzn, 2018-10-26:
Hi.
new Setup today get non VLAN interface System
I'm using now Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux is connected to switch wit access Port.

root@raspberrypi:~# plctool -i eth0 -r
eth0 00:B0:52:00:00:01 Request Version Information
root@raspberrypi:~# plctool -i eth0 -r all
eth0 FF:FF:FF:FF:FF:FF Request Version Information

The first command, I can see that first 3 blocks are right. but can't see all the devices. I'm really confused if I use the commands in the right way. but if I understand it right normally I must get all devices with plctool -i eth0 -r all?

n1000, 2018-10-27:
Hi Torsten, is the PLC device directly connected to the ethernet port? Also, can you share what adapter model it is?

n1000, 2018-10-28:
Hi Torsten, I'm submitting the FreeBSD changes, which will close this issue. Please open another one if you are still seeing issues, as it seems to be occurring on multiple platforms and is not specific to FreeBSD.

-Nate

PR #123 (2018-10-28): Add FreeBSD support, as well as fix a few bugs found while compiling in FreeBSD
Add FreeBSD support. In most cases, this was just adding a "|| defined (__FreeBSD__)" next to existing BSD related compile checks.

Add a few missing parenthesis around some macro bodies, which were noticed by the default FreeBSD clang compiler.

Switch the bpf_length to unsigned, instead of signed. The BPF man page for all supported OSes indicates it should be unsigned, and as a buffer length, unsigned makes the most sense.

Fixes #122.
closed 2018-10-28 by n1000