qca/open-plc-utils #124: QCA7000 psin, psout, psgraph not correct for QCA7000

chris-vo, 2018-10-25:
- psout man page:  frequency = 40.96 * (index + 74).
- psin man page:  frequency = (index + 74) / 40.96.
- psgraph tool seems to follow the formula of psin man page for the frequency output.

correct formula according to Qualcomm for QCA7000, QCA6410 and AR7420 is 
frequency =(index*2 + 74) / 40.96

mhei, 2018-10-26:
The formula in the man pages corresponds to the ones stated in Atheros Technical Reference Manual for FW v4.2 (INT6x00). So it seems that the tools need an addtional alternative code path for newer firmwares and/or chipsets.

maslanz, 2020-09-24:
Hi @chris-vo @mhei 

There is also decibel formula in the man pages. Is the decibel formula not correct for QCA7000, just like the frequency formula? If it is not correct, do you know the correct formula for the decibel?