qca/open-plc-utils #125: Transfer Prescaler Settings

mdkhouse, 2018-11-18:
Hi guys,

so I'm really new to this. So I'd like to apologize in the first place for asking (potentially) stupid questions...

I have six PLC adapters in my network. All based on the QCA7420 chipset, but from different vendors. It seems that the adapters from Devolo are more stable in the long run / better throughput.

So I thought it should be possible to transfer the prescaler settings from Devolo (psout/psin) to my TP-Link and TrendNet devices. In addition I'm considering upgrading the firmware on all adapters to the same v1.3.1-00-CS build. So I guess it's the same chipset after all?

Is this something you guys would recommend? Is this potentially dangerous? So I know you shouldn't mix US and EU .pibs. Are there further limitations?

Thanks!

dvlemplgk, 2018-11-19:
Prescaler settings are hardware specific and not interchangable between different hardware designs. The chipset may be the same, but the analog frontends are almost certainly different.
So just don't do that.
You shouldn't mix EU and US pibs because of regulatory issues, and that's also why you shouldn't mix prescalers of different devices.

mdkhouse, 2018-11-19:
Thanks for clarifying. Is upgrading the firmware for the chipset also problematic?

When I compare the current prescaler settings on my TP-Link device (plctool -p settings.pib MAC > psgraph settings.pib) with the latest available firmware for that TP-Link device (same revision) the .pib ps values are completely different. So when these prescaler settings are hardware specific shouldn't these prescaler values match in every firmware release?

dvlemplgk, 2018-11-19:
Manufacturers may improve prescalers with a new firmware version or they are forced to change them by regulatory changes. You'll never know.
And 2nd: with new PLC firmware version sometimes the pib changes to a new format (adding config options), so pibs are not generally interchangable even on the same device.

Upgrading the firmware shouldn't be a problem when the pib version remains the same from one fw version to another. You may try it at your own risk.

n1000, 2020-09-26:
Closing stale issues that are not directly related to a bug in open-plc-utils.

closed 2020-09-26 by n1000