qca/open-plc-utils #130: QCA7500

alexi3, 2019-10-08:
Hi!

I am using Fritz!Powerline 1220E and Fritz!Powerline 1260E under ubuntu 18.04 LTS.

CHIPSET FIRMWARE
QCA7500 MAC-QCA7500-2.5.0.37-00-20171218-CS
QCA7500 MAC-QCA7550-2.4.0.55-01-20170605-CS

plctone -h <MAC1> <MAC2>
returns
plctone: Too many carriers: Value too large for defined data type

How to fix it?

Does open-plc-utils suppport
Fritz!Powerline 1220E device?
Fritz!Powerline 1260E device?
QCA7500?
HomePlug AV2 standard?

Alexi3

HimanshuSourav, 2020-07-29:
please refer to manpages on tools usage

`man ./open-plc-utils/plc/plctone.1`

or use `-?` to get shorter list of arguments expected
```
:~/plc $ ./open-plc-utils/plc/plctone -?

 program: Qualcomm Atheros Panther/Lynx Tone Map Dump

 command: plctone [options] node peer [> stdout]

 options: [ehi:p:qsvx!?]

 -e     redirect stderr to stdout
 -h     print mean-square histogram
 -i s   host interface is (s) ["eth1"]
 -p n   coupling [0]
 -q     quiet mode
 -s     compute signal-to-noise and bits-per-carrier ratios
 -v     verbose mode
 -x     exit on error
 -!     version information
 -?     help summary
```

alexi3, 2020-07-29:
It is not a problem of tool usage, manpage doesn't help. plctone works fine with QCA7420 chip (500 MBit/s). This is a SISO chip.
QCA7500 is a MIMO chip (1200 MBit/s) using more carriers than QCA7420.
Somebody told me to use chtone, but I can't find this tool.

RvdE, 2023-02-18:
Hi,

I've just started playing around with this toolkit. My PLCs have QCA7500 chipsets and both amptone and plctone give the same error you encountered. What does work is int6ktone (with the same parameters). Maybe that helps.