qca/open-plc-utils #132: IP522CG V2 Assistance and Support

n3gwg, 2020-04-27:
open-plc-utils users and developers,

I am very new to open-plc-utils and the HomePlug environment as well. However, I am trying to learn quickly.

First let me say that I did a git clone of the source onto my Mac and with no more than typing "make" I was able to flawlessly and easily build the entire environment. Congrats on that!

I have several SerComm IP522CG V2 (200Mbps Powerline Ethernet Adapters) that I wish to program and inspect via software. Two talk to each other while the 3rd does not (I think it is reset and needs to join the group).

I have installed into my Mac an RTL USB Ethernet Port that I plugged into the one that is ostensibly not programmed. Can anyone give me any guidance on how I use these utilities or if there is a manual I can consult and read?

I guess my first interest is to be able to read the programming out of each of the devices I have. Then I'd like to be able to change parameters. I also have 2 Netgear PL1200S units, can this software program and read them as well?

Thanks in advance!

Stuart B. Tener, N3GWG
Computer Scientist

HimanshuSourav, 2020-07-29:
documentation is available in plc folder
example `man open-plc-utils/plc/plcboot.1`

n3gwg, 2022-05-24:
I read this page and it didn't really say anything about how to get it to access the IP522CG V2. Can you expound a bit on what exactly in the man page you think is applicable to this device?