qca/open-plc-utils #133: plctone + QCA7420 = "Read timeout or network error"

aldennisa15, 2020-06-13:
Versions ...
```
plcrate -!
plcrate open-plc-utils-0.0.6 OSRQCA-929 ISO C Qualcomm Atheros, San Jose CA, USA

plcrate -r -i enp2s0
enp2s0 00:B0:52:00:00:01 Request Version Information
enp2s0 04:A1:51:83:38:32 QCA7420 MAC-QCA7420-1.0.0.345-04-20120507-FINAL
```

Most plc utils work as expected, e.g. ...
```
 plcstat -m -i enp2s0
 NID DD:47:64:D3:8B:50:05 SNID 005
 STA TEI 007 MAC 04:A1:51:83:38:32 BDA 90:2B:34:D1:72:F9
 STA TEI 001 MAC 04:A1:51:9B:74:55 BDA AC:9B:0A:98:89:B8 TX 072 RX 120
 STA TEI 002 MAC A0:63:91:7B:A8:78 BDA B8:27:EB:2E:B4:5A TX 126 RX 133
 CCO TEI 003 MAC A4:2B:8C:C8:1F:76 BDA B8:27:EB:A3:8D:7A TX 103 RX 082
 STA TEI 006 MAC A0:63:91:7B:7B:5E BDA F8:63:3F:29:0B:67 TX 108 RX 079
 STA TEI 008 MAC 04:A1:51:83:3B:9D BDA B8:27:EB:41:08:7C TX 126 RX 000
```

But not working ...
```
plctone -i enp2s0 04:A1:51:83:38:32 04:A1:51:83:38:32
plctone: ToneMaps2: Read timeout or network error

plctone -i enp2s0 04:A1:51:83:38:32 04:A1:51:83:3B:9D -s
plctone: SignalToNoise2: Read timeout or network error
```
... same result for all stations on the network - they're all QCA7420 devices.

Any idea what's wrong? Can I provide any more info to assist?

Thanks

tjlusco, 2020-06-17:
I also had this issue with the QCA7420. I believe the which tone tool to use depends on the firmware version. On V1.3.0, the intk6tone tool gave numbers, but they didn't correspond with anything that made sense.

> enp2s0 04:A1:51:83:38:32 QCA7420 MAC-QCA7420-1.0.0.345-04-20120507-FINAL

The solution is to use a more recent version of the firmware, I can confirm "plctone" works with firmware versions 1.3.1 and above. This can be sourced from google ("qca7420 1.3.1 firmware"), or from TPLink firmware updates (AV600 devices, any region except US are on 1.3.1+). Just update the "nvm" file, PIBs appear to be forwards compatible. 

> \# Upload NVM Firmware (Note: this doesn't update the PIB, but a PIB is needed for it to work)
> plctool -i $PLC -P YOURPIB.pib -N MAC-7420-v1.3.1-00-CS.nvm -R $DEVICE_MAC
> sleep 20
> \# Actually Update PIB
> plctool -i $PLC -P YOURPIB.pib -R $DEVICE_MAC
> \# PLC Tone Command
> plctone $(plclist local) -h

aldennisa15, 2020-06-18:
Ah ok, thanks for that info. These are all Netgear devices and I had in my mind that there wasn't any newer firmware available for them, but I'll check again. Maybe I'll need to investigate sourcing the required firmware from elsewhere besides Netgear? Although, as one of the devices is a WiFi Access point, how successful that will be, remains to be seen... I obviously need to become familiar with nvm, pib files etc!

In the meantime, do the plc utils need to mention this firmware requirement? Either somewhere in the docs, or maybe the xxxtone utils should report a firmware error instead of a timeout, if the firmware isn't new enough? Maybe it would save a repeat question in the future.

alexi3, 2020-07-22:
I had a similar problem and changed in

open-plc-utils/ether/channel.h

#define CHANNEL_TIMEOUT 500 // 50 original

This fixed my problem.

HimanshuSourav, 2020-07-29:
`Read timeout or network error`

most probably the confirmation was not received within the specified time-limit. so the application exited with timeout error.
as @alexi3 mentioned, increasing the timeout would help in that case.

HimanshuSourav, 2020-07-29:
> I had a similar problem and changed in
> 
> open-plc-utils/ether/channel.h
> 
> #define CHANNEL_TIMEOUT 500 // 50 original
> 
> This fixed my problem.

`Read timeout or network error`

most probably the confirmation was not received within the specified time-limit. so the application exited with timeout error.
as @alexi3 mentioned, increasing the timeout would help in that case.

derekalineal, 2021-08-28:
I am receiving the same error when trying to run plctone on the QCA7000.  Has anyone had success with plctone on the QCA7000?  I tried increasing the CHANNEL_TIMEOUT as suggested by @HimanshuSourav with no change in the result.  Is there a specific firmware required to run plctone on the QCA7000 as well?  If so, do you know where I can get a newer firmware?  I'm running 1.1