qca/open-plc-utils #135: Can't write firmware to QCA7420

typhoon2099, 2020-07-29:
I recently bricked a set of TP Link powerline adapters (TL-PA4010(UK) V3) through their utility program, and can;t get plc utils to save them.

I can boot into any firmware I've tried with the PIB I downloaded from the devices before writing, but when i run:
```
plctool -i enx00e04c680044 -S FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -P header_c8eb_bedroom.pib -N FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -FF
```

I get the following error:
```
No NVM Softloader Present in Flash Memory (0x71): Device refused request
```

I can't find any mention of this error on the internet (Google returns 2 results, an unrelated issue in this repo and the line of code in the repo itself). What's the problem, and how do I fix it? As far as I understand, the firmware isn't on the device, so I can't put the firmware on the device?

HimanshuSourav, 2020-07-29:
the issue is with the command 

`plctool -i enx00e04c680044 -S FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -P header_c8eb_bedroom.pib -N FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -FF`

-S is for softloader and you are passing firmware as the argument. 
either provide the softloader file which should be available with the firmware and pib files or can try without -S i.e. remove `-S FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm`

typhoon2099, 2020-07-29:
Where can I find a softloader file? There wasn't one provided with the firmware. Can it be extracted somehow? Trying to write without the softloader causes the same error.

HimanshuSourav, 2020-07-29:
okay, if softloader is not available then please try-
`plcboot -i <<IFACE>> -N <<.nvm>> -P <<.pib>>`

are you able to query the device with a `plctool -i <<IFACE>> -Iarm`

typhoon2099, 2020-07-29:
Yup, it boots up fine, is queriable and rejoins the network (I still have two live adapters).

HimanshuSourav, 2020-07-29:
> Yup, it boots up fine, is queryable and rejoins the network (I still have two live adapters).
> 
> EDIT: Here's the output:
> 
> ```
> enx00e04c680044 00:B0:52:00:00:01 Request Version Information
> enx00e04c680044 98:48:27:63:C8:EB QCA7420 MAC-QCA7420-1.5.0.26-02-20200114-CS
> enx00e04c680044 00:B0:52:00:00:01 Fetch Device Attributes
> enx00e04c680044 98:48:27:63:C8:EB QCA7420-MAC-QCA7420-1.5.0.26-02-20200114-CS (1mb)
>         PIB 0-0 8836 bytes
>         MAC 98:48:27:63:C8:EB
>         DAK 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 (none/secret)
>         NMK 1B:14:CE:FB:A7:02:16:FD:F2:08:67:10:79:56:5D:40
>         NID 75:7F:B9:4E:09:33:07
>         Security level 0
>         NET Qualcomm Atheros Enabled Network
>         MFG tpver_401013_171025_901
>         USR tpver_401013_171025_901
>         CCo Auto
>         MDU N/A
> enx00e04c680044 00:B0:52:00:00:01 Fetch Network Information
> enx00e04c680044 98:48:27:63:C8:EB Found 1 Network(s)
> 
> source address = 98:48:27:63:C8:EB
> 
>         network->NID = 75:7F:B9:4E:09:33:07
>         network->SNID = 11
>         network->TEI = 5
>         network->ROLE = 0x00 (STA)
>         network->CCO_DA = D4:6E:0E:B9:7D:87
>         network->CCO_TEI = 1
>         network->STATIONS = 2
> 
>                 station->MAC = D4:6E:0E:B9:7D:87
>                 station->TEI = 1
>                 station->BDA = A4:4C:C8:87:DC:A2
>                 station->AvgPHYDR_TX = 292 mbps Primary
>                 station->AvgPHYDR_RX = 267 mbps Primary
> 
>                 station->MAC = D4:6E:0E:B9:C7:BD
>                 station->TEI = 2
>                 station->BDA = B8:27:EB:B1:BC:3A
>                 station->AvgPHYDR_TX = 157 mbps Primary
>                 station->AvgPHYDR_RX = 121 mbps Primary
> ```

are you able to query the bricked device? 
if it responds then you can try loading firmware using plcboot, but if no response is sent then the only way to recover would be to do a hardware reset.

typhoon2099, 2020-07-29:
No, if I query it without using `plcboot` I don't get a response. I also get some unusual flashing from the LEDs, the network LED flashes constantly, like it's trying to connect to a network. How do I do a hardware reset?

HimanshuSourav, 2020-07-29:
1. try removing the device from the power supply and plug back in. (if not tried already.)
2. try querying the device with a plctool -i <<IFACE>> -Iarm ?

if the LEDs are blinking then the device should respond to the query. 

for hardware reset there should be a procedure specified by tplink, it could range from holding a button in for some time or a hole in which pin need to be pushed. physically examine the adaptor and also can try to check the tplink manuals.

typhoon2099, 2020-07-29:
After pulling the power I now get the following:
```
enx00e04c680044 00:B0:52:00:00:01 Request Version Information
enx00e04c680044 00:B0:52:00:00:03 QCA7420 MAC-QCA7420-1.5.0.26-02-20200114-CS
enx00e04c680044 00:B0:52:00:00:01 Fetch Device Attributes
enx00e04c680044 00:B0:52:00:00:03 QCA7420-MAC-QCA7420-1.5.0.26-02-20200114-CS (1mb)
plctool: pibchain2 found bad NVM header version in device module 0
enx00e04c680044 00:B0:52:00:00:01 Fetch Network Information
enx00e04c680044 00:B0:52:00:00:03 Found 0 Network(s)

source address = 00:B0:52:00:00:03
```

HimanshuSourav, 2020-07-29:
with this state what is the issue you are seeing..? you are not able to pair the devices..?
you can restore the device to default factory settings also using `plctool -i <<IFACE>> -T`

typhoon2099, 2020-07-29:
In this current state I cannot do anything with the adapter. I can't connect to a network, and flashing fails. The only time I can do anything with the adapter is when I boot it with the NVM and PIB, which of course only works until I remove power.

Resetting to factory default settings made no difference apart from disconnecting the adapter from the network, which again is temporary until power is removed and it is rebooted with the NVM and PIB.

HimanshuSourav, 2020-07-29:
okay, could we try
load the nvm/pib using plcboot and then without powering it off use the tplink application to update the same firmware/pib.?

typhoon2099, 2020-07-29:
The update through the TP-Link Utility fails almost immediately. I tried with 5 different NVM files and 3 different PIB files.

HimanshuSourav, 2020-07-29:
okay.

the information recovered says `enx00e04c680044 00:B0:52:00:00:03 QCA7420-MAC-QCA7420-1.5.0.26-02-20200114-CS (1mb)`
could you verify this mac address with physical label on the device, is it same or different MAC on the label?

this _00:B0:52:00:00:03_ mac address is sometimes seen when we write incorrect PIB to the device.

typhoon2099, 2020-07-29:
It's a different MAC address. It has the correct MAC address when the downloaded PIB is used, but it's not present after pull the power.

typhoon2099, 2020-07-30:
I have contacted TP-Link to send the adapters back and get them fixed/replaced on the warranty. I don't know why the adapters won't write the firmware and I can't find a single reference to softloaders for the QCA4720 on the internet, so I assume it's something that you have to get directly from Atheros/Qualcom/whoever do have any hope of fixing an adapter yourself.

HimanshuSourav, 2020-07-31:
yeah, I think contacting them would be best as nothing is getting written to flash whether we use their tools or open-plc.

closed 2021-03-19 by typhoon2099

hontz1, 2021-09-30:
after plcboot, don't power off and try: 
sudo plctool -i en0 -P 4010_3.0_EN50561-3.pib -N FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -R XX:XX:XX:XX:XX:XX

XX:XX:XX:XX:XX:XX replace with mac sticker.
4010_3.0_EN50561-3.pib replace with downloaded *.pib

and firmware is in flash again.

also look at some dock here https://fitzcarraldoblog.wordpress.com/2020/07/22/updating-the-powerline-adapters-in-my-home-network/#comment-13474

netrunnereve, 2026-09-06:
> after plcboot, don't power off and try: sudo plctool -i en0 -P 4010_3.0_EN50561-3.pib -N FW-QCA7420-1.5.0.0026-02-CS-20200114.nvm -R XX:XX:XX:XX:XX:XX
>
> XX:XX:XX:XX:XX:XX replace with mac sticker. [...]

This trick worked for me. I had a bricked adapter lying around and I managed to fix it by first booting it with plcboot using the pib and nvm and then flashing it with plctool using the same pib and nvm. If you do it this way you don't need a softloader file.

By the way if you use plcboot like that your original DAK is going to get overwritten by whatever's in the pib and as per https://github.com/qca/open-plc-utils/issues/88 you can't change the DAK once you've left the bootloader. So be sure to modpib the pib first before booting and flashing it!

In my case the DAK in the original pib was set to "homeplug" (lol) by default. That wouldn't do so I had to purposely brick my adapter again by unplugging it while flashing and I then redid everything with a modified pib.