qca/open-plc-utils #136: Cloning flash?

typhoon2099, 2020-07-30:
Is it possible to read the entire flash contents of one device to put onto another device? I have bricked devices and it just occurred o me that if I could get a working device and dump the entire flash memory to flash directly onto another device I could get it up and running again to then flash the correct NVM/PIB (currently not possible as I don't have a softloader and nobody seems to have one to give me).

HimanshuSourav, 2020-07-31:
Is it possible to read the entire flash contents of one device - YES
using plcinit option o
` -o f   read all of flash memory using VS_MODULE_OPERATION`
put onto another device?  - Not with the toolkit yet. but if u can take out the flash chip and use the flash programmer -YES

but yeah sounds like a good idea though...

n1000, 2020-09-26:
Closing stale issues that are not directly related to a bug in open-plc-utils.

closed 2020-09-26 by n1000