qca/open-plc-utils #137: qca7000 Firmware versions & Firmware parser development

boraciner, 2020-09-03:
Hi everyone, I want to develop a parser for the firmware of qca7000. Attached files are extracted from 1.2.5 version which I only have as a firmware file
[files](https://drive.google.com/drive/folders/131ivjvbIQphllXZ8SprPil2G9pAJMAE1?usp=sharing)

I need to parse/understand the contents of block NVM_IMAGE_FIRMWARE do you have any idea?
Do you have different firmware versions for qca7000 ? 

Thanks!

specialistvlad, 2021-01-23:
@boraciner could you provide firmware again? Can't download it

zgb526642524, 2022-08-12:
@boraciner Currently I have firmware with version number 3.1.0

boraciner, 2022-12-06:
Can you share firmware version number 3.1.0 with me ?

boraciner, 2022-12-22:
not needed

closed 2022-12-22 by boraciner

oliviermaridat, 2022-12-22:
Hi boraciner. Did you manage the qca7000 firmware using another tool? I am curious!

boraciner, 2023-01-11:
unfortunately not

Petcho2207, 2024-08-05:
Hello everyone,

I am a university student at KMUTNB in Thailand. I am currently working on a project involving communication between an electric vehicle (EV) and a charger using Power Line Communication (PLC). For this project, I am using the QCA7000 chip, and I need a firmware version higher than v1.0.0-07. Could you please share the firmware with me?

Thank you very much.

mhei, 2024-08-05:
@Petcho2207: it is very uncool to post randomly at closed issues.
But for your information, the firmware is usally available only under NDA with QCA, but you could have a look here: https://github.com/devolo/dlan-greenphy-sdk/tree/master/QCA7000_GreenPHY_Firmware