qca/open-plc-utils #138: Unable to flash firmware: Minor Version Mismatch

arkoz, 2020-09-03:
Hello

I want to flash FW on my 4010p EU v2.3, it's working and can be queried normally :

```
enp3s0 E8:DE:27:BD:89:E7 Request Version Information
enp3s0 E8:DE:27:BD:89:E7 QCA7420 MAC-QCA7420-1.1.0.871-04-20130410-FINAL
enp3s0 E8:DE:27:BD:89:E7 Fetch Device Attributes
enp3s0 E8:DE:27:BD:89:E7 QCA7420-MAC-QCA7420-1.1.0.871-04-20130410-FINAL (1mb)
	PIB 0-0 8112 bytes
	MAC currentmac
	DAK 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00 (none/secret)
	NMK 3B:1F:0E:4B:6D:9C:90:01:2D:D9:5D:99:5A:F1:40:DF
	NID CA:B6:DB:35:F2:38:01
	Security level 0
	NET Qualcomm Atheros Enabled Network
	MFG Qualcomm Atheros HomePlug AV Device
	USR TP-LINK Product
	CCo Auto
```

I can re-flash currently installed firmware with plcboot
But when i try anything newer i'm getting message :
```
MODULE_OFFSET 0x05780000
RESERVED 0x00000000
NUM_OP_DATA 1
MOD_OP 0x11
MOD_OP_DATA_LEN 1423
RESERVED 0x00000000
MODULE_ID 0x7001
MODULE_SUB_ID 0x0000
MODULE_LENGTH 1132
MODULE_OFFSET 0x00061120
MSTATUS 0x0000
ERROR_REC_CODE 0
RESERVED 0x00000000
NUM_OP_DATA 1
MOD_OP 0x00
MOD_OP_DATA_LEN 23
RESERVED 0x00000000
MODULE_ID 0x7001
MODULE_SUB_ID 0x0000
MODULE_LENGTH 1132
MODULE_OFFSET 0x046C0000
enp3s0 00:B0:52:00:00:01 Close Session
enp3s0 E8:DE:27:BD:89:E7 Minor Version Mismatch (0x48): Device refused request
enp3s0 00:B0:52:00:00:01 Reset Device
enp3s0 E8:DE:27:BD:89:E7 Resetting ...
```

What that  `Minor Version Mismatch` exactly means? There's some version checking during nvm flash? What can be done to flash it finally?

Thanks in advance

netadair, 2021-02-15:
As far as I can tell, this message comes from incompatible PIB files.
I get the same error when trying to flash from a MAC-QCA7500-2.0.0.367-02-20140711-CS firmware on QCA7500 to anything later like MAC-QCA7550-2.2.3.32-00-20161104-CS (or MAC-QCA7550-2.10.0.32-01-20200626-CS just recently).
When extracting the pib file in my case, the old file is 20888 while newer fw's pib file length is 22344.

oraculix, 2024-09-29:
From my experience, I can confirm @netadair 's guess that this comes from incompatible PIB files.

In my particular case, I couldn't upgrade an adapter with a firmware from 2012 to a more recent one (from 2020). This worked with other adapters that had a more recent hardware revision and a more recent firmware as well.

The steps I took resolved my issue:

1. Extract a PIB from a more recent device (in my case from hw rev. 3)
2. Modify the PIB to enter the MAC of the adapter to be patched
3. Try the firmware upgrade with the new PIB.

IMHO this means:

- Either use the PIB that came with the newer firmware (but patch it with your adapter's MAC first!)
- OR use my method if you own an adapter with a more recent firmware.

Your mileage may vary and there could be hardware restrictions in your particular adapter, not allowing to upload newer – and usually larger – PIB files.