qca/open-plc-utils #139: Does INSYS Powerline GP act as a HomePlug Ethernet Adapter?

maslanz, 2020-09-16:
Hi Everyone,

I have a INSYS Powerline GP as i mentioned in the title.

So I am doing the following steps:
 - I connected the PC and the INSYS Powerline GP via ethernet.
 - I connected the PLC card which has a I2se module with the cable coming out of the INSYS Powerline GP's CP out (Pilot Out).
-  I executed the plctool -r command and the output is: 
        <ethInterface> <mac1> Request Version Information
        <ethInterface> <mac2> QCA7000 MAC-QCA7000-1.1.0.727-02-20130826-FINAL

The QCA7000 device listed in the output, represents the chipset inside the INSYS Powerline GP.

So along with these, I was waiting for one more device to be listed, if INSYS behaves like an Ethernet Adapter and i connected them truly.

I'm waiting for your help. Thank you and sorry for the English.

n1000, 2020-09-19:
Hi @maslanz , 

Have you tried appending "all" to the command? This will broadcast the version request:

plctool -r **all**

Example output:
```console
$ plctool -i em1 -r all
em1 FF:FF:FF:FF:FF:FF Request Version Information
em1 84:C9:B2:78:0A:EC INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
em1 84:C9:B2:78:0B:E6 INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
```

You can also check the network status of the locally connected (via Ethernet) device using the -m option:

```console
plctool -i XXX -m
```

This will query the local device and display all of the PLC devices that are on the same PLC network.

maslanz, 2020-09-21:
> Hi @maslanz ,
> 
> Have you tried appending "all" to the command? This will broadcast the version request:
> 
> plctool -r **all**
> 
> Example output:
> 
> ```
> $ plctool -i em1 -r all
> em1 FF:FF:FF:FF:FF:FF Request Version Information
> em1 84:C9:B2:78:0A:EC INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
> em1 84:C9:B2:78:0B:E6 INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
> ```
> 
> You can also check the network status of the locally connected (via Ethernet) device using the -m option:
> 
> ```
> plctool -i XXX -m
> ```
> 
> This will query the local device and display all of the PLC devices that are on the same PLC network.

> Hi @maslanz ,
> 
> Have you tried appending "all" to the command? This will broadcast the version request:
> 
> plctool -r **all**
> 
> Example output:
> 
> ```
> $ plctool -i em1 -r all
> em1 FF:FF:FF:FF:FF:FF Request Version Information
> em1 84:C9:B2:78:0A:EC INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
> em1 84:C9:B2:78:0B:E6 INT6400 INT6000-MAC-4-2-4203-04-4235-20100716-FINAL-B
> ```
> 
> You can also check the network status of the locally connected (via Ethernet) device using the -m option:
> 
> ```
> plctool -i XXX -m
> ```
> 
> This will query the local device and display all of the PLC devices that are on the same PLC network.

maslanz, 2020-09-21:
Hi @n1000,

Thank you for your response.

I tried the commands you suggested and my outputs are as follows.

`plctool -i em1 -r all
     em1 FF:FF:FF:FF:FF:FF Request Version Information
     em1 00:05:B6:03:3C:52 QCA7000 MAC-QCA7000-1.1.0.727-02-20130826-FINAL`

`plctool -i em1 -m
    em1 00:B0:52:00:00:01 Fetch Network Information`

So according the these outputs i think a plc network is not established.

Can I do that with a tool commands? or is the physical connection between INSYS and PLC board wrong?

maslanz, 2020-09-21:
Hi again @n1000,

Thanks for your helps. I can see the device now. I formed a network with the custom NMK and it works.

n1000, 2020-09-21:
Thanks for the update, good to hear it worked. Closing this issue.

closed 2020-09-21 by n1000