qca/open-plc-utils #143: Use Ubuntu18.07LTS system, error“program: eth1: Address family not supported by protocol”

zgb526642524, 2021-01-15:
Use Ubuntu18.07LTS system, error“program: eth1: Address family not supported by protocol”.

Ask for help, wait for your reply.

Thank you！！！

daN4cat, 2021-01-15:
Sorry but this is not a Ubuntu support forum. Search on Google.

daN4cat, 2021-01-15:
Apologies, I previously posted a comment, now deleted, under the impression this was posted in another GitHub repo. Please disregard it.

n1000, 2021-01-15:
Hi @zgb526642524, which Linux kernel are you running? Is it a customized kernel build?

I think the error you are seeing may be because the "CONFIG_PACKET" option is not enabled in your Linux kernel build. On the Linux platform, open-plc-utils uses the AF_PACKET / PF_PACKET interface for sending and receiving raw packets on the ethernet interface.

Ubuntu typically has this feature enabled by default. For reference, on my Ubuntu 20.04 box:

```console
$ grep -H CONFIG_PACKET /boot/config-$(uname -r)
/boot/config-5.8.0-38-generic:CONFIG_PACKET=y
/boot/config-5.8.0-38-generic:CONFIG_PACKET_DIAG=m
```
Can you check the output of this command on your machine to see if CONFIG_PACKET is enabled?

n1000, 2021-01-24:
Closing this issue for now. If the issue ends up being something else other than the CONFIG_PACKET kernel option, please reopen and we can take a closer look. Thanks!

closed 2021-01-24 by n1000

zgb526642524, 2022-07-24:
$ sudo grep -H CONFIG_PACKET /boot/config-$(uname -r)
grep: /boot/config-4.4.0-19041-Microsoft: No such file or directory
windows WSL  subsystem.
request assistance，Loyal thanks.