qca/open-plc-utils #145: Bin firmware

rfg81, 2021-03-03:
Hi,

I partially briked a d-link dhp-540 a long time ago. I uploaded a pib file that doesn't belong to it and d-link only makes a firmware available with the extension .bin (https://support.dlink.ca/ProductInfo.aspx?m=DHP-540). Is there a way I can flash this firmware provided by d-link with this tool? D-link firmware update tool doesn't me allow to flash with the bin file (it says it's a mismatch probably because I used the wrong pib file). The powerline kind still works, but it disconnects frequently but I'm able to use it for a short period of time. 

Thanks for any help!

bqgwzgqlhn, 2021-03-08:
You can give this utility a try: https://github.com/bqgwzgqlhn/dlink_powerline_bin_extract . It will let you extract the .pib and .nvm files from the D-Link .BIN file.

rfg81, 2021-03-08:
Thank you so much! I'm going to try it tomorrow. For flashing the files extracted, do I need a special command line to force the flash? I ask that because the d-link tool was refusing to flash the bin file due to a mismatch of the wrong PIB file I flashed before.

rfg81, 2021-03-08:
I tried to flash them using this command line but I'm not sure if this is the correct one (with the two pib files):
plcboot -i eth1 -N ar7400-v5.2.0-01-3-X-FINAL.nvm -P DHP540A2_PIB106CEB_DT.pib DHP540A2_PIB109NA_DT.pib
and I got the following error: plcboot: nvmfile2 found bad NVM header version in ar7400-v5.2.0-01-3-X-FINAL.nvm module 0
Please let me know if you want me to try something different.

n1000, 2021-03-09:
Hi @rfg81 , can you double check the PLC device version information with the following command?

```console
$ amptool -i eth1 -r
```

It looks like the two PIBs are labelled based on the region "NA" for North America, and "CEB" for "CE Class B" (usually used in EU regions). If you are in CA/US you can use the NA version of the file.

rfg81, 2021-03-09:
Thanks n1000, I got the difference after finding in another website. So I was finally able to bring back my powerline from the dead. It took a little bit of trying and error and I'm not 100% sure what really fixed it. I was able to flash the firmware using 
Power Packet Utility and factory reset it using Trendnet Powerline Utility. After that the LED were back to normal (before the power led wasn't always on when powered). It is now on firmware ar7400-v5.2.0-01-3-X-FINAL. I tried to update to a newest firmware through the softwares mentioned before by I always get an error. I'm going to leave like that for the moment. If you folks know what might be the cause for not being able to upgrade to a newest firmware, please let me know. I tried with ar7400-v5.2.4-03-5-X-FINAL.nvm and MAC-7400-v5.3.1-00-X-FINAL.nvm.