qca/open-plc-utils #146: plctone and int6ktone

alexi3, 2021-05-02:
Hi,

plctone and int6ktone have the same syntax:
plctone:   command: plctone [options] node peer [> stdout]
int6ktone: command: int6ktone [options] node peer [> stdout]

plctone sends:   Tone Map Tx Characteristics Request (0xa070)
int6ktone sends: Tone Map Rx Characteristics Request (0xa090)
(returnes other direction).
Why are differnt Request commands used?

faifa supports
0xA070 Get Tone Map Caracteristics Request
faifa does not support 0xa090

In plc/ToneMaps2.c VS_TONE_MAP_CHAR (0xA070) is used.
In plc/ToneMaps1.c VS_RX_TONE_MAP_CHAR (0xA090) is used
I think using VS_TONE_MAP_CHAR (0xA070 = Get Tone Map Caracteristics Request) in open-plc-utils plc/ToneMaps1.c would be better.

Thanks

n1000, 2021-05-02:
Hi alexi3,

The main difference between these two utilities in the version of the chipset they support / target.

The command line help string describes int6ktone as: "Qualcomm Atheros INT6x00 Tone Map Dump Utility". plctone is described as "Qualcomm Atheros Panther/Lynx Tone Map Dump". Panther/Lynx were the generation of chips that were released after the INT6x000, and include the green phy (QCA 7000) chipsets. The tonemap format and number of tones were changed, and that is the reason a new command is needed.

Thanks,
Nate

alexi3, 2021-05-03:
Hi Nate,

thanks, yes plctone and int6ktone are for different chips and use different MAC Management Header Version. plctone sends additional Coupling.

copy from wireshark:

plctone sends:
HomePlug AV protocol
    MAC Management Header
        .... ...1 = Version: 1.1 (1)
        Type: Tone Map Tx Characteristics Request (0xa070)
        Fragmentation Info: 0x0000
    Vendor MME
        OUI: 00b052
    Tone Map Tx Characteristics Request
        Reserved
        Peer address: Devolo_xx:xx:xx (00:0b:3b:xx:xx:xx)
        Tone Map slot: 0
        .... 0000 = Coupling: Primary (0)

int6ktone sends:
HomePlug AV protocol
    MAC Management Header
        .... ...0 = Version: 1.0 (0)
        Type: Tone Map Rx Characteristics Request (0xa090)
    Vendor MME
        OUI: 00b052
    Tone Map Rx Characteristics Request
        Peer address: Devolo_xx:xx:xx (00:0b:3b:xx:xx:xx)
        Tone Map slot: 0

(address replaced by xx:xx:xx)

Different number of tones is in Tone Map Tx Characteristics Confirmation (0xa071) and Tone Map Rx Characteristics Confirmation (0xa091), not in Request.

My Question is: Why other direction?
plctone: Tx
int6ktone: Rx
I think there is no reason for using other direction.

Alexi

n1000, 2021-05-07:
Hi Alexi, I'm not sure why one tool reports TX by default and the other RX... I'll leave this issue open in case someone else can comment.

alexi3, 2021-05-16:
Hi Nate,
please leave this issue open. There is another issue:
plctone node peer (without signal-to-noise and bits-per-carrier ratios) uses Tone Map Tx Characteristics Request
plctone node peer -s (with signal-to-noise and bits-per-carrier ratios) uses Tone Map Rx Characteristics Request
same tool, other direction! why?

Alexi