qca/open-plc-utils #147: MMTypes used in the demo and the MMTypes defined in the HPGP spec

tropxy, 2021-05-04:
Hi,

I have been digging a bit into the slac code and HPGP spec and I noticed the following.

Whilst in the spec the MMType CM_MNBC_SOUND.IND is defined as 0x6078 (page 518 HPGP version 1.1 Feb, 2012), in 
here https://github.com/qca/open-plc-utils/blob/bb50f635ef6cec8b52898051e0d18f4ac3fdd331/slac/evse_cm_mnbc_sound.c#L108, it is expected:
```c
CM_MNBC_SOUND | MMTYPE_IND
```
which is 
```c
0x6074 | 0x0002
```
which results in the hexadecimal value:
0x6076

And this is not the same as stated in HPGP spec.
Like this, the same happens with other MMTypes used by the Qualcomm demo. I suppose I am missing here something, but I dont know what. Can someone tell me what?

Thank you in advance!
André

oliviermaridat, 2021-05-05:
Hello @tropxy 

I can confirm a CM_MNBC_SOUND.IND is defined by 0x6076 (0x6074 | 0x0002).

And in HPGP 1.1.1 July 4, 2013, CM_MNBC_SOUND is defined as 0x6074 (in table 11-6, which means 0x6076 for MNBC_SOUND.IND).

Best regards

Olivier

tropxy, 2021-05-06:
Hi Olivier,
Thanks for your answer. I guess I need to acquire a more recent version.
offtopic: Is your business email still olivier.maridat@trialog.com?

tropxy, 2021-05-06:
And another question. When I tried to replicate for example the CM_MNBC_SOUND.IND and sniffed the package in wireshark, wireshark recognizes it as a HPGP packet but doesnt recognise it as a CM_MNBC_SOUND message. The same does NOT happen with a CM_SET_KEY. Given that wireshark has dissectors for Homeplug, I would expect wireshark to recognise the payload. Do you experience the same behaviour ?

oliviermaridat, 2021-05-06:
Hi André! Yes it is, but it is best not to publish it on Github :D Could you remove it please? Thanks!

Since at least 2.6 Wireshark is able to decode most HPGP paquets.

[image]

tropxy, 2021-05-06:
I may be doing something wrong. My wireshark recognizes a CM_SLAC_PARM but not a CM_MNBC_SOUND. I will test it again and see if I can find the error. thanks and apologies for the email issue.

tropxy, 2021-05-06:
OK, my linux machine wireshark was outdated. On my Mac I had the most recent one, but not in Linux, so now I can see the CM_MNBC_SOUND ;)

tropxy, 2021-05-06:
Could also anyone tell me if the Cipher Suit is to be used or not? According to ISO15118-3 it shall not be used, because the SECURITY_TYPE is 0x00 (as also suggested by HPGP spec), but according with Qualcomm slac, it is used:
https://github.com/qca/open-plc-utils/blob/bb50f635ef6cec8b52898051e0d18f4ac3fdd331/slac/pev_cm_slac_param.c#L126

tropxy, 2021-05-27:
Hi again, do you know what is the base MM code used for the APCM_SET_CCo message?

tropxy, 2021-06-07:
Does anyone knows? Thank you!

tropxy, 2021-06-18:
I would really appreciate if someone can provide me an answer, thanks!