qca/open-plc-utils #151: Chipset name is not reported correctly

mhei, 2021-11-29:
I have a board which is equipped with a QCA7005 chip. Now, I'm doing the following:

```
# plctool -r -v
eth1 00:B0:52:00:00:01 Request Version Information
00000000 00 B0 52 00 00 01 00 01 87 05 28 30 88 E1 00 00 ..R.......(0....
00000010 A0 00 B0 52 0E 9C 28 B9 00 00 00 00 00 00 00 00 ...R..(.........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00             ............

00000000 00 01 87 05 28 30 00 B0 52 00 00 01 88 E1 00 01 ....(0..R.......
00000010 A0 00 B0 52 00 06 0B 42 6F 6F 74 4C 6F 61 64 65 ...R...BootLoade
00000020 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 r...............
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
00000050 00 00 00 00 00 00 00 00 7C 58 1B 00 00 00 00 00 ........|X......
00000060 01 00 00 00 3F 00 00 00 00 00 00 00             ....?.......

eth1 00:B0:52:00:00:01 QCA7000 BootLoader
```

My assumption was, that the chipset name (QCA7000 in the last line) is reported correctly, i.e. I'd expect QCA7005 in the output. However, the current implementation makes some kind of mapping which always results in printing QCA7000 at this point.

vbudko, 2021-12-02:
Hi Michael,
Bootloader and Firmware knows what exact IC it is running on (QCA7000 or QCA7005), but unfortunately it is not possible to get this information via SPI interface.
For example, QCA7000/QCA7005 firmware would allow you to run QCA7005 PIB on QCA7000 IC, but not wise versa, the firmware would block running QCA7000 PIB on QCA7005 IC.
BR, Vasily Budko

mhei, 2021-12-03:
> but unfortunately it is not possible to get this information via SPI interface.

I don't understand you, see VS_SW_VER request and response MMEs above, where we get the IDENT field with chip specific content (0x007C581B). We exchange these MME via SPI connection.

But good to know, that firmware makes such checks you mentioned.

closed 2021-12-11 by mhei