qca/open-plc-utils #161: Confused about CM_SET_KEY.CNF result field

catphish, 2022-05-18:
According to the published Homeplug specifications, in the result field of CM_SET_KEY.CNF, a value of 0 is success and 1 is failure. This is further reflected in Wireshark's decoder. However. my device returns a 1 in this field, and apparently sets the key correctly. Furthermore, the code in this repository treats non-zero as a success.

https://github.com/qca/open-plc-utils/blob/358dfcf78bdaf7b0b13dcdf91cb1aae1789f2770/slac/evse_cm_set_key.c#L170

Am I misunderstanding something?

oliviermaridat, 2022-05-18:
I confirm there is a difference between the Homeplug specification and reality.
I do not know why but I would be glad to!

vbudko, 2022-05-18:
> I confirm there is a difference between the Homeplug specification and reality. I do not know why but I would be glad to!

Hi Olivier,
The Result field interpretation used by Qualcommis based on HomePlug AV 1.0 specification, where 0x01 is a success and 0x01 is a failure.
The later versions of the HomePlug specification (HomePlug AV 1.1 and HomePlug Green PHY) use a different interpretation (it considers the result field as being set to 0x00 for success and 0x01 for failure).
Hope that helps.
BR, Vasily Budko

oliviermaridat, 2022-05-18:
Thanks for your feedback Vasily.