qca/open-plc-utils #165: Long link time >4.5 seconds < 25 seconds

boraciner, 2022-12-22:
Hello everyone,
I am trying to shorten link up time, It's told that I need to receive Host action request message after setting the keys, but I don't get any message, 
After setting the keys on PEV side, I just wait until I get CPU IRQ and then start to poll link status, and It always takes above 4 seconds.
What can be wrong? I already set Background PIB HAR to 1.

Thanks in advance

oliviermaridat, 2022-12-22:
Hi boraciner,

Are you using the QCA7000 as in #137 ?
We have the same latency as yours in all our setup. Same with ST1200. I never manage to speed things up. 
It could be something in HomePlugGreenPHY protocol.
Or (if we are lucky) in QCA and ST firmwares.

I will follow this thread hoping for a better feedback ;-)

boraciner, 2023-01-11:
Hi Oliver,
sorry for late response, it was related to my code and PIB file, now It's faster. ~600ms
but.. 4 seconds after the link is up, I always get CPU IRQ :( and if I try to send some messages at that time , It's rejected (for 1 second) . 
Although I disable CPU IRQs from the enable register i still get it