qca/open-plc-utils #168: Check Signal Noise and Tonemaps

Tifoso1979, 2023-01-26:
i want to read SNR and tonemaps with the open plc utils tools. 
Is this somehow possible and what are the commands? Are examples available?

the following documentation isn't helpfull for me
https://github.com/qca/open-plc-utils/blob/master/plc/SignalToNoise2.c

Tifoso1979, 2023-01-30:
hi all,

I could find the commands to read the noise but it is not working for all chipset.

QCA7500 = OKEY
QCA7005 = no tone maps available
-> Error int6ktone: No Tone Maps Available: Operation canceled

See below the commands and error. What is here wrong? How I can read tonemaps from QCA7005? 

**__testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ plcstat -t -i eth1
 P/L NET TEI ------ MAC ------ ------ BDA ------  TX  RX CHIPSET FIRMWARE
 LOC STA 002 B8:BE:F4:71:82:19 1C:1A:DF:F8:6A:B1 n/a n/a QCA7500 MAC-QCA7550-2.13.0.30-01-20221117-CS
 REM CCO 001 B8:BE:F4:76:BE:C3 B8:BE:F4:76:BE:C2 636 581 QCA7500 MAC-QCA7550-2.8.0.30-01-20190705-CS
 REM STA 003 60:15:92:43:60:6E 60:15:92:40:70:01 009 009 QCA7005 MAC-QCA7005-1.2.5.3207-00-20180927-CS
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ 
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ 
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ 
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ 
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ int6ktone -i eth1 -s B8:BE:F4:71:82:19 B8:BE:F4:76:BE:C3
 SNR    5.072    5.219    4.440    4.898    5.167    5.057    4.976 
 ATN  -54.928  -54.781  -55.560  -55.102  -54.833  -54.943  -55.024 
 BPC    1.897    1.951    1.683    1.841    1.929    1.893    1.866 
 AGC 68 68 51 68 68 51 
 GIL 68 68 51 68 68 51 
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$ int6ktone -i eth1 -s B8:BE:F4:71:82:19 60:15:92:43:60:6E
int6ktone: No Tone Maps Available: Operation canceled
testuser@admintestuserlinux:~/open-plc-utils-master/scripts$_**

vbudko, 2023-01-30:
Hello @Tifoso1979,

QCA7005 supports only OFDM QPSK, so the modulation is always the same within all the allowed carrier frequencies.
The plctone tool is not relevant for QCA7005.

BR, Vasily

Tifoso1979, 2023-02-01:
Thank you Vasily. 

Could you tell me also more about the measurements in my example?

- SNR = what are thresholds for stable PLC communication?
- ATN = ?
- BPC = ?
- AGC = ?
- GIL = ?
- Are the values in the column history values or how I can understand this?
- Any documentation about this available?

[image]