qca/open-plc-utils #172: Can't List TPLink Devices

jamstruth, 2023-03-06:
Hello,

I've been trying to get this application working to diagnose some issues with my TPLink PA9020P Powerline devices but I can't get any kind of network topology out of the system. Is there something I've missed?

When running `plcstat -t` I get no output and running `plcstat -m` gives the following error: `plcstat: NetworkInformation2: Read timeout or network error`

vbudko, 2023-03-06:
Hi,

Why do you think Qualcomm PLC utilities should work for Broadcom based device?
Qualcomm often uses Vendor Specific MMEs.

closed 2023-03-06 by jamstruth

natevw, 2023-10-09:
Yes! I was getting all sorts of unexpected errors despite [working around](https://forum.openwrt.org/t/viewing-powerline-status-from-tp-link-wpa8630p-v2/97040/7) my VLAN setup and having had success on other routers:

> plctool: Identity2: Read timeout or network error

and

> plcstat: LinkStatistics: Read timeout or network error

Turns out that some TP-Link devices (e.g. the two I have that are combo Powerline/WiFi AP) *are* QCA-based and compatible with these tools, while others (e.g. the plain Ethernet-Powerline transceivers) are **not** compatible!

In my research I came across <https://deviwiki.com/wiki/List_of_PowerLine_Devices> (via <https://superuser.com/questions/1629114/how-does-enumerating-powerline-adapters-work#comment2488683_1629192>) which can be helpful for figuring out which chipset one might be trying to communicate with.