qca/open-plc-utils #183: No response from red beet PLC unit on ethernet

KanteshwarBondla, 2024-07-01:
I have a red beet P 2.0 Eval Board with QCA7006AQ. Now, I'm doing the following:

# sudo plctool -r -v -i eth0
eth0 00:B0:52:00:00:01 Request Version Information
00000000 00 B0 52 00 00 01 00 15 5D A0 24 D2 88 E1 00 00 ..R.....].$.....
00000010 A0 00 B0 52 22 29 4E 37 00 00 00 00 00 00 00 00 ...R")N7........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00  

My ethernet is working fine, did the updates but I am not getting any response from PLC unit. Please help me to resolve the issue.

vbudko, 2024-07-01:
Hi Kanteshwar,

Please contact CODICO using plc@codico.com
Then we can help you, thanks!

rawag12, 2025-02-18:
I have same problem.
I have TP-Link  tl-pa4010
I am trying to reconfigure it for using as v2g CCS communicator.
int6klist -ieth0 -v
Give me same result as topic starter.
No information from plc adaptor.
I am using different USB-ethernet dongle. Maybe there is a problem.
I was used WSL on windows and Ubuntu to compile plc utils and use it. Same result.